nion-software/nionswift #999: Swift 16.10.0 crashes at start up on Mac OS

kociak, 2024-01-22:
I tried to update to Swift 16.10.0, it crashes at start up on Mac OS.
I tried to create different environments I restarted my computer, updated conda ... and I consistently got the error below.

Former version (16.9.x) do work on my old environment.

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Nion Swift [3801]
Path:                  /Users/USER/*/Nion Swift.app/Contents/MacOS/Nion Swift
Identifier:            com.nion.swift.launcher
Version:                ()
Code Type:             X86-64 (Native)
Parent Process:        python3.12 [3793]
Responsible:           Terminal [898]
User ID:               503

Date/Time:             2024-01-22 10:49:00.7649 +0100
OS Version:            macOS 12.6.3 (21G419)
Report Version:        12
Bridge OS Version:     7.2 (20P3045)
Anonymous UUID:        C9F33ED4-A947-DB38-1C7C-F1FA71E13384

Time Awake Since Boot: 950 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGKILL (Code Signature Invalid))
Exception Codes:       UNKNOWN_0x32 at 0x000000010a29d000
Exception Codes:       0x0000000000000032, 0x000000010a29d000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace CODESIGNING, Code 2 

VM Region Info: 0x10a29d000 is in 0x10a29d000-0x10b3fe000;  bytes after start: 0  bytes before end: 18223103
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      __LINKEDIT                  10a1e1000-10a29d000    [  752K] r--/rwx SM=COW  ...hon3.12.dylib
--->  mapped file                 10a29d000-10b3fe000    [ 17.4M] r--/rwx SM=COW  ...t_id=69aeaedb
      GAP OF 0x7993000 BYTES
      __TEXT                      112d91000-112dfd000    [  432K] r-x/r-x SM=COW  /usr/lib/dyld

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   dyld                          	       0x112dc4601 dyld3::MachOFile::compatibleSlice(Diagnostics&, void const*, unsigned long, char const*, dyld3::Platform, bool, dyld3::GradedArchs const&) + 69
1   dyld                          	       0x112daa771 invocation function for block in dyld4::JustInTimeLoader::makeJustInTimeLoaderDisk(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, bool, unsigned int) + 82
2   dyld                          	       0x112daf80d dyld4::SyscallDelegate::withReadOnlyMappedFile(Diagnostics&, char const*, bool, void (void const*, unsigned long, bool, dyld4::FileID const&, char const*) block_pointer) const + 137
3   dyld                          	       0x112daa6f8 dyld4::JustInTimeLoader::makeJustInTimeLoaderDisk(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, bool, unsigned int) + 180
4   dyld                          	       0x112da45bc invocation function for block in dyld4::Loader::getLoader(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&) + 1557
5   dyld                          	       0x112da49e2 invocation function for block in dyld4::Loader::forEachResolvedAtPathVar(dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, dyld4::ProcessConfig::PathOverrides::Type, bool&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 207
6   dyld                          	       0x112d92db3 dyld3::MachOFile::forEachLoadCommand(Diagnostics&, void (load_command const*, bool&) block_pointer) const + 129
7   dyld                          	       0x112dccf14 dyld3::MachOAnalyzer::forEachRPath(void (char const*, bool&) block_pointer) const + 92
8   dyld                          	       0x112da39aa dyld4::Loader::forEachResolvedAtPathVar(dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, dyld4::ProcessConfig::PathOverrides::Type, bool&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 828
9   dyld                          	       0x112da359e invocation function for block in dyld4::Loader::forEachPath(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 134
10  dyld                          	       0x112d9ac87 dyld4::ProcessConfig::PathOverrides::forEachPathVariant(char const*, dyld3::Platform, bool, bool&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) const + 525
11  dyld                          	       0x112da3500 dyld4::Loader::forEachPath(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 154
12  dyld                          	       0x112da3d7a dyld4::Loader::getLoader(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&) + 860
13  dyld                          	       0x112da8b4f invocation function for block in dyld4::JustInTimeLoader::loadDependents(Diagnostics&, dyld4::RuntimeState&, dyld4::Loader::LoadOptions const&) + 427
14  dyld                          	       0x112dc3524 invocation function for block in dyld3::MachOFile::forEachDependentDylib(void (char const*, bool, bool, bool, unsigned int, unsigned int, bool&) block_pointer) const + 107
15  dyld                          	       0x112d92db3 dyld3::MachOFile::forEachLoadCommand(Diagnostics&, void (load_command const*, bool&) block_pointer) const + 129
16  dyld                          	       0x112dc339c dyld3::MachOFile::forEachDependentDylib(void (char const*, bool, bool, bool, unsigned int, unsigned int, bool&) block_pointer) const + 164
17  dyld                          	       0x112da8886 dyld4::JustInTimeLoader::loadDependents(Diagnostics&, dyld4::RuntimeState&, dyld4::Loader::LoadOptions const&) + 152
18  dyld                          	       0x112db519d invocation function for block in dyld4::APIs::dlopen_from(char const*, int, void*) + 479
19  dyld                          	       0x112d9b719 dyld4::RuntimeState::withLoadersWriteLock(void () block_pointer) + 49
20  dyld                          	       0x112db4dc6 dyld4::APIs::dlopen_from(char const*, int, void*) + 534
21  libpython3.12.dylib           	       0x109e3d895 _PyImport_LoadDynamicModuleWithSpec + 693
22  libpython3.12.dylib           	       0x109e3cbd7 _imp_create_dynamic + 167
23  libpython3.12.dylib           	       0x109cdda9a cfunction_vectorcall_FASTCALL + 106
24  libpython3.12.dylib           	       0x109deb2eb _PyEval_EvalFrameDefault + 244763
[... 287 lines cut ...]
  ]
}
}

Model: MacBookPro16,2, BootROM 1916.80.2.0.0 (iBridge: 20.16.3045.0.0,0), 4 processors, Quad-Core Intel Core i5, 2 GHz, 16 GB, SMC 
Graphics: Intel Iris Plus Graphics, Intel Iris Plus Graphics, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, LPDDR4X, 3733 MHz, Micron, MT53E1G32D4NQ-053
Memory Module: BANK 2/ChannelB-DIMM0, 8 GB, LPDDR4X, 3733 MHz, Micron, MT53E1G32D4NQ-053
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x7BF), wl0: Jul 12 2021 18:02:56 version 9.30.464.0.32.5.76 FWID 01-c081cfed
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: localIP, Ethernet, en6
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: T2Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 86.0
Thunderbolt Bus: MacBook Pro, Apple Inc., 86.0

TomaSusi, 2024-01-22:
Relatedly – I tried to install 16.9 or 16.10 in a macOS arm64 native conda environment, and I get a similar "Exception Type: EXC_BAD_ACCESS (SIGKILL (Code Signature Invalid))" error.

kociak, 2024-01-22:
@cmeyer. Running an old (Intel Core i5) processor, so can't install the arm64 version.
Just installed the packaged version (did not know about its existence!); just works like a charm on my installation.

cmeyer, 2024-01-23:
@TomaSusi the new release 0.4.23 should fix the open dialog problem. Be sure to also update `nionui` to 7.0.3.

cmeyer, 2024-01-23:
@kociak I could not reproduce the code signing problem you described. I installed a new version of macOS, installed miniconda, and used a command similar to this to install Swift:

```
conda create -n nionswift_16_10 -c nion -c conda-forge nionswift nionswift-tool -y
conda activate nionswift_16_10
nionswift
```

If you still want to pursue this issue, can you (a) try the commands above; (b) use `conda list` and paste here to show me your conda environment. If you're satisfied with the packaged download, you can just close this issue. Thanks!

kociak, 2024-01-23:
> @kociak I could not reproduce the code signing problem you described. I installed a new version of macOS, installed miniconda, and used a command similar to this to install Swift:
>
> ```
> conda create [...]
very strange. Using conda-forge in place of conda made it - no need to change OS, reinstall conda or anything else. I never used conda-forge, maybe there is a subtlety I am not aware off - just using the conda install recommended on the website.
In any case, seems to solve my problem. Are you recommending to install with conda-forge in general?

> If you still want to pursue this issue, can you (a) try the commands above; (b) use `conda list` and paste here to show me your conda environment. If you're satisfied with the packaged download, you can just close this issue. Thanks!

as a note, I can't be personnally satisfied by the packaged download. To be honest, this is really great you have made it, will help spread nionswift accross our platform users. Excellent!
But my main use of nionswift on my laptop is plug-in coding, and I understanding the external plug ins are not possible (I guess you have some sort of frozen environment there). Btw, having the possibility to add plug in to the packaged version would also be great for our users, as we have some tools implemented in the form of plug ins that they may want (example: calibration helper for cathodoluminescence data ...)

TomaSusi, 2024-01-23:
> @TomaSusi the new release 0.4.23 should fix the open dialog problem. Be sure to also update `nionui` to 7.0.3.

Thanks, confirmed this works!

cmeyer, 2024-07-25:
Please try Swift 16.11 and reopen if the failure still occurs.

closed 2024-07-25 by cmeyer