nion-software/nionswift #1010: Drawing latency increases when line profile is active

cmeyer, 2024-02-26:
The latency of a live scan increases by up to 2x when a line profile is attached to the data item. At the minimum, understand this latency (describe here) or minimize it by changing the code.

jamesrussell216, 2024-07-25:
@Ion-e can you have a look to see if we can recreate this one using the simulated cameras plz?

cmeyer, 2024-07-25:
It's still a thing and it's a complicated issue that I work on frequently.

Just to mention, it's difficult to see this on the simulator without either fixing or at least temporarily addressing this issue first.

- https://github.com/nion-software/nionswift-usim/issues/44

cmeyer, 2024-11-21:
Not seeing this anymore. If seen again, file a new issue.

closed 2024-11-21 by cmeyer

Ion-e, 2024-12-09:
Verified as fixed, thanks!