nion-software/nionswift #1028: Improve display panel title bar

cmeyer, 2024-04-15:
- make it match panel platform style
- macos: filled 'x' button on the left
- windows: title aligned left
- add a distinct 'edit' or equivalent icon button to help users find title/info editing
- consider adding size/storage info when available layout space allows it
- "(4)(512,512)(1024,) [1024MB]"

jamesrussell216, 2024-07-16:
@cmeyer quick question on this one - on your last two points why is it important to have that info in the title? Every piece of additional info that goes in there dilutes what I'd argue is the important information - the actual title. Surely having that info in the Inspector is sufficient?

cmeyer, 2024-07-16:
It already shows up as a tool-tip - but I think having the "space used" would be helpful.

cmeyer, 2024-07-17:
Something like "45MB" meaning 45MB used on disk.

If we add compression storage, we might want something like "45MB disk/90MB memory" meaning "45MB storage on disk" but "90MB when loaded in memory". I know that could be confusing, so I'm open to other ideas, but both are valid/useful pieces of information. Note: the memory usage is difficult to quantify since HDF5 files are memory mapped and don't necessarily use RAM.

An alternative to the memory memory is to have something like Firefox and Chrome have, which is a window that shows resources being used, i.e. if memory is being used.

In any case, this sounds like a different issue that shouldn't go into the title bar. I've edited the top comment to remove that part of the request.

cmeyer, 2024-07-17:
I had experimented with this, but seemingly didn't keep anything other than this minimal code snippet to experiment with:

```
diff --cc nion/swift/Panel.py
index a8aa52c3,a8aa52c3..9c50c3b3
--- a/nion/swift/Panel.py
+++ b/nion/swift/Panel.py
@@@ -440,7 -440,7 +440,7 @@@ class HeaderCanvasItem(CanvasItem.Canva
                  drawing_context.text_align = 'center'
                  drawing_context.text_baseline = 'bottom'
                  drawing_context.fill_style = '#000'
--                drawing_context.fill_text(self.title, canvas_size.width // 2, canvas_size.height - self.__text_offset)
++                drawing_context.fill_text(self.title + " \N{LOWER RIGHT PENCIL}", canvas_size.width // 2, canvas_size.height - self.__text_offset)
```

jamesrussell216, 2024-08-12:
Notes:

@Tiomat85 has put in a PR for the edit functionality so hopefully we can tick that off.

Other items on here are styling questions - we will have some trouble doing the MacOS one as we don't have macs! How do we want to handle this from an Issue admin point of view? We can go and move the title to be left aligned (@cmeyer - is that just for Windows or MacOS as well? I'm guessing the former but want to check) we can do the Windows side but you might need to pick up the MacOS side!

jamesrussell216, 2025-03-06:
@cmeyer was just discussing this one with @Tiomat85 and hadn't realised that the edit title functionality is based off a double-click off the title - it took us about fifteen mins to find it when we knew the functionality should be there! Not quite sure we're ticking the box of helping users find title/info editing, was there a reason we didn't make it icon based in the end?

cmeyer, 2025-03-06:
No reason - I just was doing the preliminary work in https://github.com/nion-software/nionswift/pull/1328. It is relatively easy to change the layout and functionality of the header now that it is done with canvas items. So... a PR against this issue would be welcome.

Tiomat85, 2025-03-21:
As a USER I want to be able to, and easily identify how to edit the title of a Display Item.

Currently the way to do this is to double-click on the title, and a micro-inspector opens (covering the title bar) which is not easily intuitive for new users nor completely functional since it covers it.

So:

- Add a new edit button to the left hand side of the close button on the title bar (right side of it on macos).
- The edit button shall be a small button, similar size to the Close button, with the icon of a pencil.
- Edit button opens the existing microinspector window.
- Microinspector window lowered ~20 pixels so it sits below the title bar.

cmeyer, 2025-03-21:
Those changes sound good. Hopefully the changes in #1328 make these changes easier.

Tiomat85, 2025-03-27:
@jamesrussell216 @cmeyer 
Spec creep question.  What do we want to happen if you press the edit button while the microinspector is currently open. We have 3 options:

1. Nothing
2. Close the window as if Cancel were pressed.
3. Close the window as if Done were pressed.

1 is obviously easiest, but 3 is probably what I would prefer as a user.

jamesrussell216, 2025-03-27:
@Tiomat85 I'd vote #3.

cmeyer, 2025-03-27:
I have no strong opinion. I can see the case for (1) or (3), but definitely NOT (2).

cmeyer, 2025-03-28:
I realize I put my general comments in the PR and not here... so here goes (slightly updated):

Overall, I continue to dislike the 'edit' button. The standard for editing titles in other applications is mouse-over with single- or double- click. I'd rather improve the highlighting of the mouse over, perhaps including a cursor change, than using up title bar space for an edit button.

On macOS Pages application:

[image]

I like this better because it doesn't take up extra title bar real estate except during mouse over; and it's very simple: a down arrow for discovery and the dialog opens like ours does and there is much less chance of mistakenly clicking close when you mean to click edit.

cmeyer, 2025-10-16:
I think any change is going to look odd for a little while. FWIW Adobe Premiere and Blender both use left aligned titles and both have similar "panel" functionality. VS Code makes the left-aligned title appear in a tab-like control. I can't find another Windows application where the panel titles are centered.