nion-software/nionswift #1029: Add mask tools needed for 4D STEM imaging

cmeyer, 2024-04-15:
Note: this is an umbrella issue. The effort for this issue is listed as "a day" since the work is done in sub-issues.

ndellby, 2024-04-15:
Chris,
This ties into another issue with 4DSTEM masks, much of the stuff one wants to use these for is relative to the current BF disc (the image of the aperture). E.g., an ABF mask is an annulus with an outer diameter equal to the BF disc and an inner diameter 1/3 of that.
So to make previously designed masks re-usable one would want to have a means of scalable masks.
Niklas

-Bruker Confidential-

**EXTERNAL EMAIL**

### Tasks

- [ ] https://github.com/nion-software/nionswift/issues/1027

cmeyer, 2024-04-15:
I added a to-do item for scaling in the task list above; but let's talk offline to fill in the details.