nion-software/nionswift #1034: Allow user to customize the scale marker (style, position, font, etc.)

cmeyer, 2024-04-24:
Email Mar 29, 2023: "I have a question about image scale bars in swift - is there any way to move them around so that they don't block the gold ball, and remove the '60kV SuperScan'?"

Tiomat85, 2025-04-14:
From internal discussion - To be broken down into sub-issues with differing levels of priority.

Adjustable default for the software, which new display items have as standard.  The preferences menu contains these default options so we can adjust it.

Once created, each Display Item can then have their options individually adjustable.  For each Display Item, the Inspector should contain these options for adjustment.  This may include the mini-inspector from double-click.

Options a User can adjust:
Position - Bottom Left, Bottom Middle?, Bottom Right
Background - To resolve the problem when drawing white on white, that can handle opacity.
Toggle Voltage
Toggle Detector Name

Options NOT in scope:
Adjusting font itself (family, colour, scale) and style
Freeform movement of the scalebar
Adjusting the Size/Thickness

Future Developments:
Scalebar style (in-image, outside-image, graphlike axes)
Draw Calibrated Grid
Exports matching - WYSIWYG

cmeyer, 2025-09-03:
Thanks! For the sake of moving this along, I made a number of minor modifications, put this behind a feature flag, and merged it. Let's file individual issues against these changes moving forward.

Changes:
- added scale marker text color, since without it light backgrounds are useless
- standardized naming `scale_marker_...`
- fixed numerous whitespace and formatting issues
- fixed scale marker text drawing to match old drawing behavior, vis-à-vis line spacing and padding
- changed inspector to look decent on macOS, shortened label fields
- put everything behind feature flag until additional steps are complete. see #1647

```
diff --git a/nion/swift/ImageCanvasItem.py b/nion/swift/ImageCanvasItem.py
index 62808bfe..1d413ae7 100644
--- a/nion/swift/ImageCanvasItem.py
+++ b/nion/swift/ImageCanvasItem.py
@@ -242,14 +242,16 @@ class ScaleMarkerCanvasItemComposer(CanvasItem.BaseComposer):
                  info_text: str, screen_pixel_per_image_pixel: typing.Optional[float],
                  get_font_metrics_fn: typing.Callable[[str, str], UISettings.FontMetrics],
                  scale_marker_position: str | None,
-                 background_fill_color: str | None) -> None:
+                 scale_marker_text_color: str | None,
+                 scale_marker_background_color: str | None) -> None:
         super().__init__(canvas_item, layout_sizing, cache)
         self.__dimensional_calibration = dimensional_calibration
         self.__info_text = info_text
         self.__screen_pixel_per_image_pixel = screen_pixel_per_image_pixel
         self.__get_font_metrics_fn = get_font_metrics_fn
         self.__scale_marker_position = scale_marker_position
-        self.__background_fill_color = background_fill_color
+        self.__scale_marker_text_color = scale_marker_text_color
+        self.__scale_marker_background_color = scale_marker_background_color
 
     def _repaint(self, drawing_context: DrawingContext.DrawingContext, canvas_bounds: Geometry.IntRect, composer_cache: CanvasItem.ComposerCache) -> None:
         dimensional_calibration = self.__dimensional_calibration
@@ -268,7 +270,7 @@ class ScaleMarkerCanvasItemComposer(CanvasItem.BaseComposer):
 
             text1 = dimensional_calibration.convert_to_calibrated_size_str(scale_marker_image_width)
             fm1 = get_font_metrics_fn(scale_marker_font, text1)
-            text_height = float(fm1.height-4)
+            text_height = fm1.height
             text2 = info_text
             fm2 = get_font_metrics_fn(scale_marker_font, text2)
             text_width = max(fm1.width, fm2.width)
@@ -284,13 +286,13 @@ class ScaleMarkerCanvasItemComposer(CanvasItem.BaseComposer):
             with drawing_context.saver():
                 drawing_context.translate(canvas_bounds.left + origin_x, canvas_bounds.top)
                 baseline = canvas_bounds.height
-                total_text_height = text_height * 2 + padding
+                total_text_height = text_height * 2
                 total_height = scale_marker_height + padding + total_text_height
                 background_top = baseline - total_height
                 drawing_context.begin_path()
                 drawing_context.rect(0 - padding, background_top, content_width + padding * 2, total_height + padding)
                 drawing_context.close_path()
-                drawing_context.fill_style = self.__background_fill_color
+                drawing_context.fill_style = self.__scale_marker_background_color
 
                 drawing_context.fill()
 
@@ -305,20 +307,21 @@ class ScaleMarkerCanvasItemComposer(CanvasItem.BaseComposer):
                 drawing_context.line_to(bar_left + scale_marker_calculated_width, baseline - scale_marker_height)
                 drawing_context.line_to(bar_left, baseline - scale_marker_height)
                 drawing_context.close_path()
-                drawing_context.fill_style = "#448"
+                drawing_context.fill_style = "darkslateblue"
                 drawing_context.fill()
                 drawing_context.font = scale_marker_font
-                drawing_context.stroke_style = "#000"
+                drawing_context.text_baseline = "bottom"
+                drawing_context.stroke_style = "black"
[... 175 lines cut ...]
         self.__image_display_handler = DisplayTypeChooserHandler(display_item, document_controller)
+
         widget = Declarative.DeclarativeWidget(document_controller.ui, document_controller.event_loop,
                                                self.__image_display_handler)
-        scalebar_widget = Declarative.DeclarativeWidget(document_controller.ui, document_controller.event_loop,
-                                               self.__scale_marker_options_handler)
+
         self.add_widget_to_content(widget)
-        self.add_widget_to_content(scalebar_widget)
+
+        if Feature.FeatureManager().is_feature_enabled("feature.scale_marker_customization"):
+            scale_marker_widget = Declarative.DeclarativeWidget(document_controller.ui, document_controller.event_loop, ScaleMarkerOptionsHandler(display_item, document_controller))
+            self.add_widget_to_content(scale_marker_widget)
+
 
 class LegendPositionChooserHandler(Declarative.Handler):
     def __init__(self, display_item: DisplayItem.DisplayItem, document_controller: DocumentController.DocumentController) -> None:
@@ -4837,3 +4857,6 @@ class DeclarativeColorChooserConstructor:
             return widget
 
         return None
+
+
+Feature.FeatureManager().add_feature(Feature.Feature("feature.scale_marker_customization", "Scale marker customization (position, color, background).", False))
```

cmeyer, 2025-09-03:
Merged in 811f627d78214b4f8fd38c57cb5f4a6c69c20450. Closing this PR. Please review note above.

cmeyer, 2025-09-03:
Oops reopening. I put the last two comments on the wrong issue. Doh!