nion-software/nionswift #1037: Additional improvements to SVG export: no padding, only specify height for fixed aspect ratio items, etc.

cmeyer, 2024-04-30:
```[tasklist]
### Tasks
- [ ] If a single image is being exported, only require user to specify height
- [ ] Allow height to be specified in inches, centimeters, or pixels
- [ ] Display final size in specified units according to height and aspect ratio
- [ ] Allow font size to be changed
```

cmeyer, 2024-08-08:
I merged the initial work by @lisham2000. I think we should add a resolution (ppi) when exporting to inches or cm (#1051).

I also added the following to the to-do list:

- Use min size of 3" and max of 12" (or equivalent in cm) for initial values
- Remember choice of inches/cm/pixels.

jamesrussell216, 2024-11-20:
Hi @cmeyer, not urgent but I talked to @lisham2000 this morning and he said we were happy with where with this now pending potential further work in the future - is this issue OK to close now?

cmeyer, 2024-11-21:
Sure I can close this with the additional note that #1126 is still open and desired, but not urgent.

closed 2024-11-21 by cmeyer