nion-software/nionswift #1040: Deleting several Data items whilst using right click shows Data items incorrectly as "Display items"

Ion-e, 2024-05-02:
there is an inconsistency behaviour between deleting one data item vs deleting more using the right click submenu.

[image]

jamesrussell216, 2024-07-16:
Sorry @Ion-e can you be more specific here as to what the issue is? What is the inconsistency?

Ion-e, 2025-03-21:
I would say that keeping the naming consistent, as long as they make sense for the user would be the best thing to do. Currently I don't understand enough about the distinction to make a decision, so happy to have this ticket closed.

But I would add that if the distinction between display item and data item is not relevant to the user then it would make sense to have the naming consistent.

closed 2025-03-21 by cmeyer

Ion-e, 2025-03-24:
Has this been completed or set as invalid? @cmeyer