nion-software/nionswift #1041: Some of the editable units are not actually Editable

Ion-e, 2024-05-03:
[image]
Inconsistency behaviour in editable units in edit boxes.

Actual behaviour:  some of the editable units are not actually editable and will default to it's original value.

Expected behaviour: edit boxes and units should behave consistently throughtout the software to avoid misinterpretations by the users.

[image]

jamesrussell216, 2024-07-16:
@Ion-e can you do us an exact list of which are the problem edit boxes? I can't tell from the screenshots alone which ones those are?

cmeyer, 2025-03-20:
@Ion-e  Yes, please file specific issues in either `nionswift` project or `nionswift-instrumentation-kit`. Close this issue?

closed 2025-04-10 by Ion-e