nion-software/nionswift #1049: Graphics on 4D data show incorrect position in centered mode

Brow71189, 2024-05-15:
If you put a graphic (ellipse in my example) in the center of a data item containing 4D data, and switch the displayed coordinates of the graphic to either "Pixels (centered)" or "Fractional (centered)", the position of the graphic in the inspector will not be (0, 0), but almost the same as with "top left" coordinates. It doesn't show the exact same position as in top-left display mode, but definitely not the correct centered one.
On a 2D data item it works as expected.

cmeyer, 2024-05-15:
Ellipse position set to 0,0; data is (4,4),(32,32); display set to 'Pixels (Center)'
[image]

cmeyer, 2025-03-25:
I'm no longer able to reproduce this issue. If it comes up again, please re-open.

closed 2025-03-25 by cmeyer