nion-software/nionswift #1051: If a single image is being exported, only require user to specify height

cmeyer, 2024-05-20:


cmeyer, 2024-05-31:
If we only ask for height, it is extremely difficult to get multiple 2D image SVG exports with different aspect ratios to match in resolution and I think having a matching resolution between exports is a desired use case in addition to the use case of having the same height for multiple exports.

I think we should go back to something where the user can specify width, height, or resolution, with the other two being dependent on whichever one the user chooses to specify. Something like this, where two of the values are dimmed (auto calculated) depending on the last value the user entered.

[image]

bentps, 2024-05-31:
I'm a little unclear about what resolution means in the context of vector format like SVG. My understanding (happy to be corrected) is that the practical implication of the resolution parameter in this context is determining the relative size of the font with respect to the image dimensions. Is that correct?

cmeyer, 2024-05-31:
For 2D images, when drawing to SVG, we draw all of the pixels. As @ndellby pointed out, one option might be to just leave it at that. A typical SVG viewer draws pixels at 96dpi, so a 1024x1024 would display as 10.66in x 10.66in. A 256x256 would display as 2.66in x 2.66in. And a 96x96 image would display as 1in x 1in.

However, we can into problems when we add graphics, text, and the scale marker.

Lines will typically draw at 1 pixel on the screen and text at something like 14pt in the UI of Swift. So a rectangle on either a 256x256 or 1024x1024 image (or any other image) will draw at 1 pixel in the UI. Likewise, text will always draw as 14pt.

So when we export to SVG, if we draw the image at its pixel size, then the lines and text will draw relative to their pixel size. A 256x256 image will have text that is "14pt" or takes up 96dpi x 14px / 256px = 5.25 "dots per inch of the screen" or (5.5%) of the height of the image. A 1024x1024 image will have text that is "14pt" or takes up 96x14/1024 = 1.31 "dots per inch of the screen" (1.4%) of the height of the image.

In some cases this might be acceptable, but in most cases one will want to have control of how large the image appears by default and separately how large lines and text appear on the image, since usually exporting to SVG is done for a specific use case in other presentation software such as a web browser or a word processor.

To this end, I propose that we want to allow the user to specify how large the image will appear when imported into other applications. This can be specified in three main ways:
- a fixed height (width is determined by the data aspect ratio)
- a fixed width (height is determined by the data aspect ratio)
- dpi (width/height are determined by dividing the data dimensions by the dpi)

If you want to export different data items such that they have a uniform height, you would choose a fixed height. A 256x256 or 1024x1024 would be drawn with all pixels, but appear exactly 4in tall. Assuming 96dpi display, the font would be drawn to be 14 pixels _independent of data size_.

If you want to export different data items such that they have height proportional to their data size, you would choose a fixed DPI. A 256x256 at 96dpi would draw to display in 2.66in height and a 1024x1024 @ 96dpi at 10.66in height. However, the text would still be 14 pixels _independent of data size_.

In the future, I propose that we also want to specify the font size.

Also note: "14pt" usually refers to 1/72 in. in print. I'm intentionally not being too rigorous about the difference between "14pt" and "14px". In browsers, 1pt = 1/96 seems standard, so interchangeable with 1pt = 1px.

bentps, 2024-05-31:
Thanks Chris. 

Summarizing our chat conversation, I think we agreed that rather than these options:

-     a fixed height (width is determined by the data aspect ratio)
-     a fixed width (height is determined by the data aspect ratio)
-     dpi (width/height are determined by dividing the data dimensions by the dpi)

We should have:

-     a fixed height (width is determined by the data aspect ratio)
-     a fixed width (height is determined by the data aspect ratio)
-     a fixed ratio between physical units and figure size (height/width are determined by choosing e.g. nm/inch)

My opinion is that getting a working version of fixed height and width export out the door is a higher priority than working out the kinks of the third option.

In any of the above cases, it would be nice if the user could select the font size relative to the figure size.

jamesrussell216, 2024-06-10:
Quick note post discussions last week; @lisham2000 to implement as @bentps has described in the previous comment, however changing the font size will be covered under work done on the parent issue #1037

Ion-e, 2024-08-08:
Verified as fixed.

Thanks!

cmeyer, 2024-08-08:
This is NOT fixed. One part of it has been fixed. This key point is still open:

> User may want to export 2D multiple data items such that they have match resolution.

cmeyer, 2024-08-08:
After discussion, I made a separate issue for the remaining item and will re-close this issue.
- https://github.com/nion-software/nionswift/issues/1126

closed 2024-08-08 by cmeyer