nion-software/nionswift #1068: Minor cleanup to export dialog

cmeyer, 2024-06-12:
More items to-do (after merge):

```[tasklist]
### Tasks
- [ ] elide the directory string to fit to width. the UserInterface.py truncate_string_to_width may be helpful.
- [ ] clean up the prefix string to auto-enable if user enters prefix string
- [ ] save the other preferences from viewmodel. the UserInterface.StringPersistentModel and similar classes may be helpful here.
```

bentps, 2024-07-02:
Is this the right place to comment on the (generally excellent) recent changes to the export dialog?

There are 4 checkboxes for what info to include in the file name. In the previous version, the state of the checkboxes was remembered from one use of the dialog to the next. Now, all four checkboxes are checked each time the dialog is opened. I much prefer the previous behavior.

As an aside, I only ever use the "include title" box. The other ones seem unnecesary to me, but I don't know if anyone else uses them.

cmeyer, 2024-07-02:
Yes, this is a good place to make minor comments. If something is more significant or you want it tracked separately, you can also file a separate issue and I will link it from here depending on its significance.

The "remember the choices" item is already listed above as the 3rd task. I reworded it to make it more evident.

It's difficult to know what people use; maybe users can comment if they're following this issue. Otherwise, I don't like to remove functionality. I think remembering the options will go a long way towards addressing the "some people may find some options extraneous" issue.