nion-software/nionswift #1081: Eliminate display type choice (dictated by data type)

cmeyer, 2024-06-25:


bentps, 2024-07-02:
What would be the default display type for shifts of an length N sequence of 2D images, which is N x 2?

cmeyer, 2024-07-02:
Whatever behavior happens now would be the default (displays as image). In addition, in the short term, I would propose adding a special "redimension" processing: "redimension to stack of 2 spectra of length N", similar to how one can redimension it now to be "redimension to sequence of. spectra of length N", which would display as a line plot by default.

Overall, I would also propose the following:
- redimension to stack, as described above
- allow option for line plots showing sequences/1d-collections to show them stacked
- implement some of the features proposed in "[future data items](https://github.com/nion-software/nionswift/wiki/Future-Data-Items)" page, particularly vector-valued data arrays that display as line plot or whatever.

If you want different behavior, let me know. I think the key item is the last one (vector valued arrays), but also the most difficult.

bentps, 2024-07-02:
Shifts for sequences of 2D images are a special case that deserves special handling. Currently Andreas experimental align image sequence automatically sets the display type for the shifted output to be a line plot. That has been a huge quality of life improvement to the previous behavior where the shifts were shown as an image, and I would be unhappy if that behavior went away.

When I do need to use the multidimensional processing menu measure shifts, that shift data is shown as an image, and then I used the display type dropdown in the inspector to change it to a line plot display. I would prefer if it was automatically shown as a line plot, but at the very least I don't want to lose an easily available option to display it that way. A menu item to redimension the data sounds like a downgrade from the current workflow.

cmeyer, 2024-07-02:
Good point. I added the issue below and made this issue dependent on it.
- https://github.com/nion-software/nionswift/issues/1087