nion-software/nionswift #1087: Ensure that output from align sequence is displayed as stacked line plots

cmeyer, 2024-07-02:


jamesrussell216, 2024-07-05:
@Tiomat85 can you put your explanation you just gave me of this in here please?!

cmeyer, 2024-07-10:
I updated the main description of the bug. Is that sufficient?

Tiomat85, 2024-07-10:
What Chris has put there is a far better description than you would have got from me.  I like the idea of a display hint going along with the data, as somewhere in the software should have some context as to what the data actually represents.

jamesrussell216, 2024-07-10:
Yes thank you both - that's a big help, I actually understand the problem now!

Picking up on what @Tiomat85 said @cmeyer - we have that in the metadata somewhere at the moment? Like a property that is data item "type" or are most things just inferred of dims?

jamesrussell216, 2024-07-19:
@cmeyer nudge

jamesrussell216, 2024-07-24:
@cmeyer nudge