nion-software/nionswift #1089: Making Inspector more useful - Collapsed by default, maintaining state

Tiomat85, 2024-07-03:
In the spirit of looking at the Inspector as a whole, can I propose that on Swift launch all Inspector sections are collapsed by default rather than expanded, and we store by section name whether it is currently expanded/collapsed.  If the sections are well organised and named it should be easy to find what you are trying to find, and clicking to expand a section (where you can see all section names because of not having everything expanded) is less hassle than needing to scroll 3 pages down.

_Originally posted by @Tiomat85 in https://github.com/nion-software/nion-internal/discussions/121#discussioncomment-9937235_

This issue, if resolved, may also supersede Issue #66.

cmeyer, 2024-07-03:
I don't think this is a good idea. See comments in the discussion.

Ion-e, 2024-07-17:
Issue verified as fixed. 

@cmeyer thoughts on #66? Also, this issue needs closing :)

closed 2024-07-17 by jamesrussell216