nion-software/nionswift #1104: Splitting and closing panels in workspaces do not work consistently and leave drawing anomalies

cmeyer, 2024-07-19:
The drawing anomalies are likely the result of the recent layout algorithm changes.

From @bentps:

> Example 1:
> 
> Start with a blank workspace. Use the toolbar button to divide the display into 2x2 panels. Select one of the smaller panels, and again divide that 2x2. The current behavior is that the sizes of the original 2x2 division get radically changed by subdividing one of the panels. The behavior I'd prefer, and what mostly used to happen, is that a panel that gets divided does not change the size of any surrounding panel.
> 
> Example 2:
> 
> In some cases, adjusting the size of a display panel by clicking and dragging on the panel edge will also affect other nearby panels. My preferred behavior would be that if I am dragging a panel edge, only that panel edge is affected.

### Steps to reproduce layout issue:

Make a 2x2 workspace using toolbar, select bottom right panel, split into 4x4 using toolbar.

[image]

### Steps to reproduce drawing anomaly:

Make a 3x3 workspace using toolbar, select bottom right panel, split into 4x4 using toolbar. This will lead to a wacky workspace, but this drawing problem occurs with or without the fix for the wacky workspace. Now start closing the new 4x4 using top-right of bottom-right 3x3 panels. Drawing problem occurs quickly.

[image]

cmeyer, 2024-07-26:
Fixed b17966a766aa883ea1944069217d443c98bc8b9b 6b6e892c2e336eb6ede852581379a517590d44fb

closed 2024-07-26 by cmeyer

Ion-e, 2024-08-05:
Issue verified as fixed.

Thanks!