nion-software/nionswift #1112: Need the ability to lock images so they can't be deleted

cmeyer, 2024-07-27:


jamesrussell216, 2024-08-07:
@cmeyer would it be better to lock data items from delete, or have an "Are you sure?" message box confirmation? Or recycle bin functionality for that matter?

cmeyer, 2024-08-07:
I marked this as 'needs planning' so that we can discuss it. Also, for reference, see #123.

I personally dislike "Are you sure?" messages - and it doesn't fit nicely into the UX of Swift (although we might use them already in a few places). Generally, UX best practices dictates to use "undo" rather than "Are you sure?"

I don't think undo (or "are you sure?") accomplishes what I want, which is to make it challenging to delete my original items in the demo library I use.

I do like the idea of a trash/recycle-bin functionality and I think we should do that no matter what we decide here. The notes in #123 need some work, but the main idea is there.

Also, I'm mostly concerned about "base" data items, meaning original data that can't be replaced, as opposed to computed data which can usually be recreated without a huge data loss (sometimes the computation parameters are important though...).

What about making original items (i.e. un-computed data items) be locked by default. To delete them permanently you would need to explicitly unlock them in the inspector. All other items are unlocked by default and deleting them immediately deletes them. User can explicitly lock if desired. Adding the trash-recycle-bin functionality means that any data item can be moved to the recycle bin without any restriction; but emptying trash requires unlocking any locked items or skipping them.

jamesrussell216, 2024-08-12:
I think the challenge here is your user story as described there is to make it difficult to do something - which is essentially promoting a potentially frustrating UX; by design. Is that the best user story for this issue? It's the sort of thing I'd usually suggest talking to a couple of core users about explicitly. For instance are there use cases where someone might want to easily delete acquired data - such rubbish preliminary images - that we'd accidentally create a very annoying problem for?

cmeyer, 2024-08-12:
I'm open to many different ideas, but the bottom line is that it is too easy to accidentally delete key data with no recourse. We can put this issue on hold until we've had a chance to hash out the details.

cmeyer, 2024-08-14:
It mostly works if (a) you do it right away, immediately after deleting; (b) you're not acquiring data at the time; (c) there isn't a bug.