nion-software/nionswift #1121: Inconsistent Behaviour on Graphics Tools

Tiomat85, 2024-08-06:
The position and shape options for the varying Graphics tools are inconsistent, with some being able to be resized, some not, some following the 'lock' rules, some not.

|  |   | Movable | Resizable | Respects Position Lock | Respects Shape Lock |   |  
|-- | -- | -- | -- | -- | -- | -- | --
|Line Tool |   | Y | Y | N | Y |   |  
|Rectangle |   | Y | Y | Y* | Y |   | Can still rotate no matter what lock.    Position can be changed by dragging the resize markers in the corners
|Line Profile |   | Y | Y | N | Y |   |  
|Spot Tool |   | Y | Y | Y* | Y |   | Can still rotate
|Ring Tool |   | N/A | Y | N/A | N |   |  
|Ellipse Tool |   | Y | Y | Y | N |   | Can still rotate. Position can be changed by dragging the resize markers   in the corners
|Cross Tool |   | Y | N | Y | ? |   |  
|Interval Tool |   | Y | Y | N | Y |   |  
|Wedge Tool |   | N/A | Y | N/A | N |   |  
|Lattice Tool |   | ? | ? | ? | ? |   |  

What I would expect is that all tools should have consistent following of the tools and lock options:
All 'free' tools are translatable unless the Position lock is engaged.
All tools are resizable/adjustable unless the Shape lock is engaged.
All 'free' tools are rotatable unless the Rotate lock is engaged (1d Interval excluded)

Additionally it would make sense for the 'grab' points to be for a specific option. Grabbing the corner points (normally used for resizing/reshaping) and it being a move is inconsistent with itself. 
So I would propose:
Grab points at the extremes, used for resizing.
Grab point in the middle (or just dragging anywhere inside the bounding rectangle) for translation.
Grab outside of the bounding box and moving for rotation.

cmeyer, 2024-08-07:
Additional references:
- #961
- #149
- #58

cmeyer, 2024-08-07:
Can you make add a task list and make the to-do items more explicit? It's difficult to interpret the combination of the table and broad statements like "grab points at extremes" - what does that mean for a point or lattice?

I'd be open to a 'rotation lock' attribute and associated UI. And I'd like to see the line-type graphics resizable from the center. If there are other particular inconsistencies, let's list those, discuss, and once agreed upon, convert into issues and get PR's for them.

Tiomat85, 2024-08-08:
Sure, I can try break it down into individual tasks.  Initially I started with just the table to try document the current behaviour to identify inconsistencies/bugs for individual graphics items.   It was only once I had that together and while testing it that there were some behaviours/UX that had inconsistencies that I could not easily put in table form.

cmeyer, 2025-03-20:
@Tiomat85 Can we close this and/or follow-up by filing specific issues?

jamesrussell216, 2025-05-21:
I've had a look through these and picking out specific things I think we should consider fixing...

For clarity, whilst considering this I've worked to the definitions of 
position = object centroid
shape = length & width of object

Additionally many of these get simpler post https://github.com/nion-software/nionswift/issues/1401

- [ ] The line graphic doesn't respect position lock. I would think this needs to either fix the centre of the line (equivalent to pressing ALT) similar to the rectangle tool (centroid is fixed, but you can still change the shape)
- [ ] Ring tool. These are constrained to be position locked (you can't move them from the image centre) so I think that should be either reflected in the position lock settings or that should be removed for this item.
- [ ] Ring tool, shape lock doesn't lock the annular radii, I would think it should.
- [ ] Point tool, shape has no meaning here and probably shouldn't appear.
- [ ] Interval tool, position lock should lock the centroid (i.e. be equivalent to ALT press at present).
- [ ] Wedge tool. These are constrained to be position locked (you can't move them from the image centre) so I think that should be either reflected in the position lock settings or that should be removed for this item.
- [ ] Wedge tool, shape lock does nothing - it should lock the angles I would think.
- [ ] Lattice tool, shape locks both position and shape at the moment. Position lock should lock the lattice periodicity. Shape should lock the size of the circles.

jamesrussell216, 2025-05-21:
@cmeyer does that work for you? I didn't think it was worth splitting each of these out into separate issues given the amount of overlap here.

cmeyer, 2025-05-21:
- https://github.com/nion-software/nionswift/issues/248
- ring tool MUST be centered on the origin for it to work as a Fourier filter
- ring tool shape lock: agreed
- point tool: it's way more work to get rid of the lock checkbox than just leave it. is it worth it?
- interval: agreed
- wedge: MUST be centered on origin. again: is it worth the effort?
- wedge angles: yes
- lattice: agreed

Tiomat85, 2025-05-23:
I would say since there are quite a few instances there of individual bits maybe not being worth it, would the sum of them all mean it would be worth considering?

A quick look makes me think it might not be that complicated, just a bit time consuming.

Graphic - Extend to include 'has_position', 'has_shape' properties that define whether a graphic has those attributes.  As a default, have those return true (all graphics have all attributes). I don't think these will need to be persistent as it is type based not instance based.

For each graphic type, if it does not have an attribute, override the getter of that specific attribute and make it return false.

In the inspector add bindings on the checkboxes for 'enabled' linking to those has_X properties.  Would also be good to add a tooltip if disabled to indicate "This graphic does not have this attribute" or somesuch.

jamesrussell216, 2025-05-23:
My view is exactly that - there's a lot of little bits like this in Swift that on their own are small and insignificant. The problem is the quantity of them adds up to the point it effects the software usability. And the only way we progress it is by chipping away at them.