nion-software/nionswift #1129: The "Rotation" for "Crop" in "Multi_Dimensional Processing" does not work

PhysXu, 2024-08-09:
Hi,

When I tried to crop my HAADF and corresponding SI data, I found the "Rotation" function for "Crop" in "Multi_Dimensional Processing" does not work. Please see the snapshot below:
[image]

The cropped region on the right is not from the crop square on the left, after I rotated the crop square.

The version of my nionswift is 16.11.0
[image]

I remembered that this function works well in the last version of nionswift (I updated my nionswift two weeks ago).

Is it easy to be fixed?

Or how can I go back to an older version of Nionswift?

cmeyer, 2024-08-09:
Unfortunately, the multi-dimension crop has never supported rotation. We're in the process of moving most of the multidimensional code to the standard Swift processing menu. You can follow that progress here:
- https://github.com/nion-software/nionswift/issues/430

I'm also adding a note to that to-do item for this specific functionality ("Ensure rotated crop can be performed [efficiently] on sequences/collections"). I'm closing this issue in favor of the one above.

I don't have a short-term solution other than writing your own script. If you decide to do that, you'll have to write a general function using something like the code here: https://github.com/nion-software/niondata/blob/c4b96fdd84e07e45e217d61527a18a443e6c5703/nion/data/Core.py#L1018 (essentially using `scipy.ndimage.map_coordinates` in some type of short console script (Ctrl+K) like below, where r05 is your display item, (assigned with Ctrl+Shift+K).

```
coords = ... (see link above)
show(scipy.ndimage.map_coordinates(r05.data, coords))
```

closed 2024-08-09 by cmeyer

PhysXu, 2024-08-10:
> Unfortunately, the multi-dimension crop has never supported rotation. We're in the process of moving most of the multidimensional code to the standard Swift processing menu. You can follow that progress here:
> 
> * [Add pixel by pixel processing to map 0d, 1d and 2d operations to collections/sequences #430](https://github.com/nion-software/nionswift/issues/430)
> 
> I'm also adding a note to that to-do item for this specific functionality ("Ensure rotated crop can be performed [efficiently] on sequences/collections"). I'm closing this issue in favor of the one above.
> 
> I don't have a short-term solution other than writing your own script. If you decide to do that, you'll have to write a general function using something like the code here: https://github.com/nion-software/niondata/blob/c4b96fdd84e07e45e217d61527a18a443e6c5703/nion/data/Core.py#L1018 (essentially using `scipy.ndimage.map_coordinates` in some type of short console script (Ctrl+K) like below, where r05 is your display item, (assigned with Ctrl+Shift+K).
> 
> ```
> coords = ... (see link above)
> show(scipy.ndimage.map_coordinates(r05.data, coords))
> ```

Dear Chris, thanks very much for your reply and the suggestions.
I will try my best by following your suggestions.