nion-software/nionswift #1134: Exporting single data items is more difficult now

jamesrussell216, 2024-08-15:
From @cmeyer ...

-

The main feature I want is to be able to export and name my file. Doing that for a single data item is not an obvious experience and leaves the export dialog in a weird state: uncheck all of the boxes, and enter your filename in the "prefix".

I propose three menu items (for now):

Export Item... (works on a single item, uses title of item, displays error message if fails)
Export Items... (works on multiple items, uses dialog for naming, displays log of success/failures [always])
Export SVG... (unchanged, for now)

2nd item could also be "Batch Export..."

Is it perfect? Probably not. Does it accomplish what we want? Probably. Let's do it and move on.

--

jamesrussell216, 2024-08-15:
Some minor suggests UI-wise just to hopefully make things a bit clearer...

For the file menu and right-click context menus (so same functionality for both)

- A single "Export..." menu item which expands to the three sub menu items suggested above...
      - "Single Item..."
            - File save menu dialog, default name as item title. Displays error message if fails.
      - "Batch Export..."
            - New multi-export window
      - "Export SVG..."
            - Existing SVG panel

Behaviour...

If a user has only one item selected, allow all three sub menu items (so that you can still access the batch export windows for a single item export, as originally requested in https://github.com/nion-software/nionswift/issues/932 but also do a quick export if desired)

If a user has multiple items selected, disable the "Single Item..." option as this doesn't make sense in this context.

cmeyer, 2024-08-15:
This sounds good!

cmeyer, 2024-10-04:
Fix applied 3e084a57873f79184354498fbf111c1d04f7c9fd

cmeyer, 2024-10-08:
@jamesrussell216 can we close this?