nion-software/nionswift #1135: Bug with Swift max/minimising behaviour

jamesrussell216, 2024-08-15:
Lay over from this issue: https://github.com/nion-software/nionswift/issues/940

Repro case from @Ion-e 

Screen 1 >start with swift maximised.
Drag Swift to screen 2 and drop at the top corner of the screen (This should move swift to screen 2 and maximise it in screen 2).
Minimise Swift.
Reopen Swift.
Swift will reopen in window 1.

The original ticket was closed as the vast majority of the problems with this had been resolved, but this lower priority case was still behaving badly so we decided to split it out into a second, lower priority issue

cmeyer, 2025-03-20:
I don't think we have much control over this as this behavior is controlled by Qt and the operating system.

How do other applications behave?

If it is a Qt problem, is it worth filing an upstream bug? Or close this issue?

Ion-e, 2025-03-21:
Other applications tend to restore back to the last used position.

Can't answer the other questions, sorry!

Tiomat85, 2025-03-24:
It might be worth a verification that it is Qt, as there were a few similar ones I did a while back were related to uitool saving/restoring state at the wrong times.  It may be missing a save state on maximise in Window 2 or something.

jamesrussell216, 2025-03-26:
OK after my misunderstanding; this is Qt standard behaviour - my bad. I'm going close this then as "working as intended", even if I think it's a bit weird!

closed 2025-03-26 by jamesrussell216