nion-software/nionswift #1150: Compression of large HDF5 data

TomaSusi, 2024-09-03:
I understand that Swift stores all multidimensional data such as 4D-STEM datasets we're collecting on the ARINA as uncompressed HDF5 files. 

I recently took a few datasets of total ~38 GB in size, and zipped them on my desktop to send to a collaborator... resulting in an archive that is 2.7 GB. While such extreme ratios may not be possible simply by enabling fast lossless HDF5 compression, but some compression really would be vital to keep our data volumes manageable.

In my opinion, doing 4D-STEM on the ARINA using Nion Swift is not going to be viable on the long term for most labs if we waste so much space by not using any compression. (Even better would be to use Zarr but that's another story ;)

cmeyer, 2024-09-03:
Quick response (I'll add more to this later): we're already tracking this internally. I'll track down the internal issues and make them public, possibly in this issue and possibly in a new issue.

cmeyer, 2025-03-20:
See also
- https://github.com/nion-software/nionswift/issues/1264