nion-software/nionswift #1157: Bug - Inspector: graphics exander has the "Display" placed unintuitively

Ion-e, 2024-09-13:
Currently at the bottom of the graphics expander and lack of indicator to indicate users that the display affects all graphics.

Please consider any of the following suggestions or a combination of any:
1. Implementation at the top of the Graphics expander.
2. Different indentation.
3. Add a separation line.

@cmeyer

jamesrussell216, 2024-09-16:
I would think moving it to the top would be more than sufficient to resolve the problem. Probably need to give the indentations of the graphics objects a once over anyway but I'm fairly sure there's another issue for that.

KRLango, 2024-09-19:
There are several sections of the inspector that have a "Display" drop-down. All of them have it at the bottom of that section. In terms of consistency - having them all in the same place is good.

My second point here is that the primary aim of the graphics section is the graphics. The current calibration of the numbers feels secondary to me, and therefore under the graphics feels like the right place for it. But I do realise that there is nothing to indicate that it relates to all of them.