nion-software/nionswift #1166: Feature Request - copy image to clipboard

SeanK59, 2024-09-18:
Ctrl+v to copy selected image from display panel to clipboard.

cmeyer, 2024-09-23:
What are you trying to accomplish?

Edit > Copy already copies an SVG version of the display panel and you can paste into applications that accept the 'image/svg+xml' mime type, although the Qt backend seems to have a bug on at least macOS preventing this from working properly due to clipboard permissions or something.

Also, the menu item Processing > Duplicate duplicates a data item.

If you're asking to make a version of the image in JPEG form, copy has the same questions as exporting to JPEG - what resolution do you use for the image? You may not follow that discussion, but this becomes an issue if there are graphics on the image or if it is a line plot.

cmeyer, 2024-12-31:
Closing this issue unless more information about what is being requested is provided. Please re-open if you want to provide more details!

closed 2024-12-31 by cmeyer