nion-software/nionswift #1183: Bug - Colour Picker: The Stroke Width is locking up the tool

Ion-e, 2024-10-02:
The issue appears inconsistently.

When adding a thick Stroke width, there seems to be a lag with the drawing.

When removing the thickness, nothing happens.

Ion-e, 2024-10-03:
There are no outputs being printed.

However I have noticed a pattern, when you have 2 layers, If you change the Stroke width of layer 1, nothing will happen *until* you press enter on the stroke width of layer 2.

Ion-e, 2024-10-03:
After restarting Swift, now it seems to be working fine, so haven't been able to reproduce it further.

cmeyer, 2024-10-04:
This bug probably occurs because the inspector does not check whether the value has changed after `editing_finished` on any particular field and enters a cycle where the line width and the next field selected loop changes.

I think this issue will go away after #1093 which will use declarative and handle this situation correctly.

See also #1186.

Ion-e, 2024-10-04:
To be tested after declarative inspector merge.

Ion-e, 2024-10-22:
PR that fixes this has been merged. > https://github.com/nion-software/nionswift/pull/1201

Ion-e, 2024-10-31:
Issue veryfied as fixed, tool works flawlessly now, thanks!

closed 2024-10-31 by Ion-e