nion-software/nionswift #1186: Bug - Colour picker: Several edit boxes appear as selected

Ion-e, 2024-10-03:
It's possible to select different edit boxes. Only the last selected box will change when typing, so not a major issue.

[image]

cmeyer, 2024-10-04:
This bug can be reproduced by putting keyboard focus in an inspector field like the line plot layers stroke width and then tabbing out. The stroke width will remain highlighted.

This bug occurs because the inspector does not check whether the value has changed after `editing_finished` on any particular field and calls `select_all` on the field _after_ losing focus.

The best way to fix this is to complete #1093 which will use declarative and handle this situation correctly.

See also #1183.

Ion-e, 2024-10-04:
To be tested once Declarative inspector merge.
It can also be triggered by clicking in different edit boxes.

Ion-e, 2024-10-22:
PR that fixes this has been merged.  > https://github.com/nion-software/nionswift/pull/1201

Ion-e, 2024-10-31:
Verified as fixed, looks great now, thanks!