nion-software/nionswift #1193: Export error File name is incorrect

Ion-e, 2024-10-07:
[image]

Fail dialog shows the attempted filename instead of the failed data/display item name.

cmeyer, 2024-10-07:
With the current naming scheme in the application, it might make sense to give both the title and the attempted file name.

Ion-e, 2024-10-08:
Perhaps, at least the user *should know* which data item was not exported. For reference: https://github.com/nion-software/nionswift/issues/1193

cmeyer, 2025-03-21:
@Ion-e Can you revisit this? I think this is fixed.

jamesrussell216, 2025-03-26:
Need to be careful here as "Multiple Data Items" don't have a "Title" - not entirely sure what's going in the code with these in how they get named, so just a note for whoever picks this up they might have to take care with those!

Ion-e, 2025-03-26:
[image]
An additional user case where this may be confusing. If  a user unticks the name (and other options) they may end up with a situation where some of their exports have failed and they cannot locate which ones.

KRLango, 2025-03-28:
This is covered by #1398

Ion-e, 2025-03-28:
Looks good to me.

closed 2025-04-01 by KRLango

Ion-e, 2025-04-02:
Verified as implemented.