nion-software/nionswift #1204: Histogram doesn't populate when on composite line plots

Ion-e, 2024-10-14:
As title and image.
Tested on https://github.com/nion-software/nionswift/pull/1201 however it may happen on master too.

[image]

jamesrussell216, 2024-10-14:
@Ion-e I'm not sure it should - there isn't really contrast per se on a line plot.

Ion-e, 2024-10-22:
That it's a very valid comment, however:
[image]

cmeyer, 2025-03-21:
@Ion-e close this as unplanned?

Ion-e, 2025-03-21:
Closing as per invalid.

closed 2025-03-21 by Ion-e

cmeyer, 2025-03-21:
Invalid but I decided to add some documentation addressing this issue and other recent line plot layer name changes.
4a15da05f7e0993c152f265b4b680f7ca363b35d

Ion-e, 2025-03-24:
Excellent, thank you!

jamesrussell216, 2025-03-25:
Apologies for commenting on a closed thread but I just want to check I understand here - so the histogram tool works fine for all single 1D data items, it just doesn't work for "Multiple Data Items"?