nion-software/nionswift #1205: Inspector: Fill colour doesn't seem to work correctly

Ion-e, 2024-10-14:
When adding composite line plots, the second line plot fill colour appears as Blank.

Ion-e, 2024-10-14:
Added to PR

jamesrussell216, 2024-10-14:
@Ion-e is this a bug or intended?

Ion-e, 2024-10-22:
That wouldn't be a question for me but for @cmeyer.  I cannot see a use case for it being working as intended.
Behaviour still persists.
I'm happy if this is not in the Kanban board, however I wouldn't removed the "Bug" type until we have more info on the behaviour.

cmeyer, 2024-10-22:
The intended behavior is that only the first layer is filled. Subsequent layers are stroked, but not filled. The idea is to reduce clutter with the subsequent layers.

Are you reporting something other than this behavior?

Ion-e, 2024-10-23:
> The intended behavior is that only the first layer is filled. Subsequent layers are stroked, but not filled. The idea is to reduce clutter with the subsequent layers.
> 
> Are you reporting something other than this behavior?

This ticket is about that behaviour only. Other issues will be reported in different tickets. 

Closing issue as per comment, thanks!

closed 2024-10-23 by Ion-e