nion-software/nionswift #1210: Inspector: Graphics section is not being removed when removing graphics

Ion-e, 2024-10-14:
As title.

STR: Select a display item > Add a graphic (or drag some line plots if on line plot graph) > Select the same display item again > Observe the "Graphics" section gets added to the inspector > Control Z > Graphic should disappear < Graphic section remains unchanged. (To actually remove the graphics section you need to "manually refresh".)

Ion-e, 2024-12-10:
Update: the particular section of the deleted graphic gets removed, however the "graphics section" in the inspector remains.

cmeyer, 2025-03-21:
@Ion-e I can't reproduce this anymore. Recheck?

Ion-e, 2025-03-27:
Issue verified as failed.

[image]