nion-software/nionswift #1212: Stroke width on line plots stops updating correctly

KRLango, 2024-10-14:
If you have a line plot with multiple layers and swap the order of the layers by pressing the move layer up button on layer 1 (making it layer 0). The stroke width will no longer update when changing the number in the edit box for layer 0 (was layer 1). If you update anything else about the appearance of the plot, for example changing layer 1's stroke width, then all updates will happen, including the stroke width for layer 0.

cmeyer, 2024-10-15:
This sounds like a good one to write a test for to show the issue...!

cmeyer, 2024-10-16:
This works ok for me with the #1201 and my new `Inspector.py` changes. Confirm?

jamesrussell216, 2024-10-17:
@cmeyer I looked at this today and it's still happening for me as described above. Quick test case...

1. Using the uSim EELS Camera, take two snapshots (hereafter referred to as Snapshot 1 and Snapshot 2).
2. Drag Snapshot 2 data item onto the axis of Snapshot 1 to make a "Multiple Data Items" display item.
3. In the Inspector, change the Layer 1 Stroke Width to 5 and press return.
4. (observe the line width of the red Layer 1 graph increase in thickness)
5. Press the little "up arrow" underneath Layer 1 to promote it to Layer 0.
6. Change the (now) Layer 0 Stroke Width to 1 and press return.
7. (observe the line width not changing thickness).

KRLango, 2024-10-18:
I have double checked using #1201 and it is still showing the same behaviour for me.

cmeyer, 2024-10-23:
PR
- #1226

Ion-e, 2024-10-24:
Excellent, sending to test until it's been merged.

Ion-e, 2024-10-29:
Issue verified as fixed, thanks!