nion-software/nionswift #1220: Data info property is unable to handle 4d data

lisham2000, 2024-10-22:
The data info property of DisplayItem.py is unable to handle 4d data as the rank of displayed data shape does not necessarily equal the rank of the dimensional calibrations

lisham2000, 2024-10-22:
@cmeyer im assuming this just needs to be a change in calibrated_dimensional_calibrations property but wanted to check before i changed it that im not missing where this is already done

lisham2000, 2024-10-22:
Think this is fixed in a line on #1192  but might be worth a check

cmeyer, 2024-10-22:
This is slightly messy since there is also a special case for spectrum images. I made a PR which I think addresses the issues but we still might find some corner cases in the future.

Check out the PR and let's go from there. #1222

cmeyer, 2024-10-22:
... also, as mentioned in #1192, these changes should go into their own PR as it is mostly independent from the SVG code. Once we merge #1222, can you revisit #1192 and remove the `DisplayItem.py` changes in that PR? If there are still changes needed in `DisplayItem.py`, either comment here or file another issue. That way we keep the underlying changes required for SVG export separate from the SVG export dialog itself.

commit c542aa49 (2024-10-23): Fix #1220. Ensure data_info works with spectrum images, 4d data, and sequences of 4d.
closed 2024-10-23 by cmeyer