nion-software/nionswift #1233: Deleting "Parent" data items

Ion-e, 2024-10-24:
There is an inconsistency on how Swift handles the deletion of a "parent" data item and how it affects the daughters. For images, it seems that deleting a parent deletes the whole branch of daughters (at least with certain processings) whilst the daughter line plots(Specifically referring to the composite line plots items called multiple data items) remain (possibly because the composite line plots, when created, are created as a new data/display item).

cmeyer, 2024-10-24:
We use a heuristic to determine what to delete. Dependent computed data items and any associated displays are deleted if the parent is the last remaining input to a computation. Line plots, on the other hand, are displays. The heuristic for displays is the underlying data item is only deleted when the last display for that data item is deleted. In the case of a composite line plot, there will often be another display of the underlying data, i.e. the single item display. If you delete that first, then delete the composite, the data item will get deleted.

So, I would mark this as intended behavior, even if a little confusing at times.

This heuristic may warrant a larger discussion, but let's put that off to the future as there is a lot of complicated scenarios to think through that will come up with the work on acquisition objects anyway.

Ion-e, 2024-10-31:
That makes sense.

It's very clear, thanks for the explanation! :)

Closing as per comment and resolution.

closed 2024-10-31 by Ion-e