nion-software/nionswift #1240: Collection of issues with composite Line plot

Ion-e, 2024-10-31:
1. When creating a line composite plot, the created item is not automatically selected, instead it seems like the next item in the data panel list is selected.
2. If adding a line plot with a specific colour scheme to the usim EELS camera line plot (to create a composite line plot), if chosen a specific colour for the line plot it won't persists. (I can see how that may be as per designed for "default" colour line plots to not have all blue stroke/fill colours).
3. There is no way to identify the "parent items" in a line composite plot, at least I couldn't find any.

jamesrussell216, 2025-01-27:
@Ion-e can you reframe this one around the first bug as discussed please!

Ion-e, 2025-01-27:
Issues 1 and 3 have been made into 2 new bugs, this is now a dupe.

closed 2025-01-27 by Ion-e