nion-software/nionswift #1279: Line profiles have a small visual nudge

Ion-e, 2024-12-09:
Choose a 3k by 3k Generated data item
Create a line profile that covers most of the item
Observe the issue

[image]

cmeyer, 2024-12-11:
A few notes: this looks like a line, not a line profile. Assuming it is a line, can you give me the exact end coordinates of the line (using the inspector) and the width in pixels of the viewing area? You'll need to use a screen ruler [perhaps like this](https://www.tomsguide.com/how-to/how-to-use-windows-11-powertoys-screen-ruler).

Ion-e, 2024-12-13:
It's a line profile. Please find attached a similar case, and a copy of the data item where it has been observed :)
[image]