nion-software/nionswift #1282: "Reveal in data panel" option when right clicking on item in the data panel

Ion-e, 2024-12-10:
Redundant button: if you are already in the data panel and right click on an data item the option of "Reveal in the data panel" will appear, which will have no useful functionality. Any reason why this was left here?

[image]

jamesrussell216, 2024-12-12:
Not sure what to do with this one to be honest. Obviously, completely agree with the premise of the issue - it is a bit of a daft context entry for this context. But I'm not sure if it would be worth removing that specific example versus having a wider discussion about what functionality is actually wanted in these right click context menus? Presumably a Export SVG (where appropriate) option would also make sense in here?

Ion-e, 2024-12-12:
Without knowledge of the original design specifications I would consider this a consistency issue if wanted to be considered an issue.

cmeyer, 2025-03-21:
Let's just close this one. It's a consistency issue but the special case to handle it is going to be more difficult to maintain than is worth it.

closed 2025-03-21 by cmeyer