nion-software/nionswift #1284: Generated Data item created but failed to populate with information

Ion-e, 2024-12-10:
Although this is a minor issue as we are using "artificial data", I would like to make the topic of this issue the fact that there are no warnings whatsoever and wonder if this happens on acquisitions or processings, if the user gets a more clear feedback that it has failed.

[image]
[image]

cmeyer, 2024-12-11:
Steps to reproduce?

Ion-e, 2025-01-27:
Sending to test until repro case found.

Ion-e, 2025-01-31:
STR: Use the folowing parameters.

[image]

cmeyer, 2025-01-31:
That's an invalid combination - which we should probably warn about, but this is very low priority and a non-trivial amount of work.

As an explanation: the '1+1' data type is produces a collection of spectra that is displayed as a line plot. So because it is already a collection, you can't make a collection of a collection right now due to the limited capabilities of the data descriptors.

Note there is a related draft PR and several related issues, although the underlying problem is not directly expressed well in issues.
- https://github.com/nion-software/nionswift/pull/1182
- https://github.com/nion-software/nionswift/issues/1087
- https://github.com/nion-software/nionswift/issues/1081
- https://github.com/nion-software/nionswift/issues/618

Ion-e, 2025-02-03:
Closing as per comment.

Thanks for reviewing!

closed 2025-02-03 by Ion-e