nion-software/nionswift #1285: Adding new Line plot display layers populate with info

Ion-e, 2024-12-10:
Generate a 1D line plot display.
In the inspector, add a new layer 
Observe the created layer will also have populated data in the line plot.

Is this what we would expect from this functionality?

jamesrussell216, 2024-12-12:
Is this the "Line Profile" feature @Ion-e ?!. I've had a look at it and there seem to be quite a few weird bugs in here that I found without even trying (just trying to understand the feature!). Just jotting a quick list for prosperity...

- Line profiles seem to have a default fill colour which can give some odd displays. Not sure if that's intended?
[image]

- Typing text into the fill colour edit text box crashes the legend.

Otherwise I would guess this is the expected behaviour but I'm not sure on the use case for it.

Ion-e, 2024-12-12:
Unsure on the issues you raise. Also, I would suspect that both line plots should behave similarly if not the same, regardless of how they are created.

The original issue is triggered by creating a new layer. However when adding a fill colour to that layer (which should be empty as there shouldn't be any data on that layer) and adding fill colour, the 1D line plot gets populated with the colour.

[image]

cmeyer, 2025-03-21:
Adding a layer is only useful if there is another layer to display. There are a few cases for this, but the primary one is an image shaped like 1200x3, then the 'row' for each layer can be 0, 1, or 2. This is also covered (minimally) in the user's guide under 'Line Plot Layers'.

See also:
- #589

closed 2025-03-21 by cmeyer