nion-software/nionswift #1288: Swift slow down/partial freeze

Ion-e, 2024-12-10:
Go to "Open project".
Close the window without changing anything.
Attempt to select a data item.
Observe the software becomes laggy for a few seconds.

jamesrussell216, 2024-12-20:
@Ion-e I tried to recreate this one but couldn't - does it still happen for you?

Ion-e, 2024-12-20:
It does, 100% repro.

jamesrussell216, 2025-01-27:
OK I've seen this now, but only the first time pressing the Open Project button - so literally...

- Open Swift
- Go to Open Project
- Close Open Project window
- Click on data item. Lag is a few seconds worth.

Does that tie in with what you have @Ion-e or is it all the time for you?

Ion-e, 2025-01-27:
Hmm, so out of 10 times, it has been reproduced 9 times.

jamesrussell216, 2025-01-27:
Without closing Swift in between??

Ion-e, 2025-01-27:
Without closing Swift in between. Uncertain why we are seeing different behaviours.