nion-software/nionswift #1289: Interval graphics are able to be taken out of the range of the 1d plot

Ion-e, 2024-12-13:
Specifically to 1D plot graphics as these items are not able to be zoomed in or out.

this type of graphic should have a hard limit

jamesrussell216, 2025-01-27:
@Ion-e I'm not sure I understand this one from what's written here?

Ion-e, 2025-01-27:
Take a line profile graphic and drag it out of the display item completely. > The linre profile graphic will go completely out of the range of the 1D graph.

cmeyer, 2025-01-27:
It's not the ideal user experience, but this is intentional behavior as it allows measuring along an edge without changing the shape of the line profile graphic. I'm open to other ideas of how to handle it.

Ion-e, 2025-01-28:
I would say a blocker so that the Graphic doesn't extend over the line graph would be something I would go for. Whether you would want the block to happen at the beggining of the line profile or after the end, it's not up for me to decide.

jamesrussell216, 2025-01-29:
Could you limit it so the mid point of the graphic can't go beyond the limit? That way you can stop it flying off the plot but still allow easy measurement over the edge?

cmeyer, 2025-01-29:
Adding limits to graphics is complicated to implement due to the huge number of side cases, e.g. user enters value in inspector, what happens? A good workaround for this is: don't do that. Does there need to be more?

jamesrussell216, 2025-02-03:
Well in that I would expect the edit box to be able to detect an invalid input and handle it appropriately as it would similar, in this case snapping to the upper limit would probably be fine. Yes though there certainly is a point in this case of maybe this isn't a high priority item, but software usability is a slippery slope; what's an obvious "well just don't do that" for one experienced user is done accidentally by a new one, causing frustration.

cmeyer, 2025-03-21:
As stated above, this is intentional behavior. I don't see anything to change here. I'm not planning on fixing this.

cmeyer, 2025-03-21:
I think I misunderstood this - you're referring to the interval, not the line profile graphic itself.