nion-software/nionswift #1295: Add ability to switch collection directly from the data panel

cmeyer, 2024-12-31:
The Collections panel allows the user to change collections; and the Data Panel shows the currently selected collection. Sometimes the user will not have the Collections panel open and the extra line in the Data Panel provides a good place to put a UI element to be able to switch the collection without the Collections panel being open - either a down arrow triggering a pop-up menu on the right; switching collections when the user clicks on the collection name, or a pop-up menu when the user clicks anywhere in the collection name section.

[image]

In the image above, the collection area is the area with the text "Persistent (0)".

jamesrussell216, 2025-05-15:
@cmeyer seems like a potential easy win to just change that to a dropdown? I think that would be a more intuitive UI interaction that a swapping mechanism.

Ion-e, 2025-06-17:
-- Test plan:
- Open the collections and the data panel.
- Create several groups with different names in “File > Add group”.
- Create a few data items, live items and distribute them amongst those groups.
- Double click on the bar under the “Data panel”.
- A combobox menu should appear.
- The list of all the groups should appear.
- Click on Live.
	-The data panel should change to only live data items.
- Iterate this process throughout the different groups.
	- The data panel should update to reflect the chosen group.
- Click on the “Live” group in the collections panel.
	- The data panel should update with the “Live” group.
	- The combobox should update to show “Live”.
- Click the combobox.
- Hover the mouse over a different group as the chosen one.
- Press scape.
	- The combobox should close, and remain on the “Live” group.
- Create another group “New group”.
- The group should now appear in the combobox.

closed 2025-06-25 by cmeyer

Ion-e, 2025-07-18:
Verified as implemented. Thanks!

commit 85e8a558 (2025-06-25): Fix #1295. Add combo box in place of the collection title for easier collection swapping.