nion-software/nionswift #1300: It's possible to create 2 workspaces with the same name

Ion-e, 2025-01-09:
As title.

jamesrussell216, 2025-01-27:
@Ion-e is that necessarily a problem? I agree it might not be the most sensible thing to do, but...

Ion-e, 2025-01-27:
Unsure how the users handle this feature so wouldn't be able to say. Bug I reckon is still valid, although I would agree not very high priority and can be taken off the board.

closed 2025-03-21 by cmeyer

cmeyer, 2025-03-24:
I think this must be one that I accidentally closed without commenting - but I would close it with a "not planned" unless there is a big reason to keep it open.

jamesrussell216, 2025-03-26:
No worries. I agree with "Closed as not planned" is fine; previously I've worked on a loose principle of if something is definitely not going to get done inside of the next 12 months to close it as not planned, just to manage ticket quantity. I think there's relatively limited value in keeping an issue hanging around if it's not going to get done in that sort of timeframe but I know not everyone agrees with that view!!!