nion-software/nionswift #1307: Generate data: 2d scan of 1d spectrum generates a 2d image

Ion-e, 2025-01-13:
As title, shouldn't a 2d of a 1d spectrum should be created instead?

cmeyer, 2025-01-13:
This works as expected. The display of a spectrum image, i.e. (16x16)x(512x1) is a raster image formed by taking a range of energies and displaying the integration of those energies as a 2d image. And there are no controls on the image. The energy axis can be "picked" using the pick tool and the "pick" will display the 1d data summed within the pick region.

See this issue for a suggestion to allow alternative "slices" of the data.
- #669

An alternate slicing of a spectrum image would be to display the spectrum with controls for the position. This would be of limited use since the user almost always wants to see a sum of spectra in a region specified on the collection (raster) display.

Ion-e, 2025-01-14:
Closing bug as per comment.

Thanks!

closed 2025-01-14 by Ion-e