nion-software/nionswift #1310: Control Z doesn't capture Grid/Thumnail views

Ion-e, 2025-01-17:
STR:
1. Have more than 1 data items and select a display item.
2. Show as "Grid view".
3. Double click on a different data item.
4. Display item should change to the selected one.
5. Press Control Z
Actual:     Display panel reverts to the "Display item" view of the previous display item.
Expected: Display panel reverts to the "Grid browser" view of the previous display item.

jamesrussell216, 2025-03-26:
@cmeyer quick question on this one; both behaviours (that @Ion-e references; Actual and Expected) are currently present. As described above CTRL-Z doesn't return to the Grid Browser. However if you change a display panel to Grid Browser, then change it to Thumbnail view, then CTRL-Z it will go back to Grid Browser. Question is which behaviour do we want, I'm assuming it's CTRL-Z consistently returning to the Grid Browser in both cases but I thought I ought to check?

cmeyer, 2025-03-26:
Undo was never intended to track changes to views - only changes to the underlying 'data model'. There may be cases where the view changes in response to an undo, but that is a side effect of the underlying data model changes being reverted.

Overall, undo has a lot of deep issues like "is it possible to undo an acquisition"?