nion-software/nionswift #1312: Doing "Alt + click and drag from line graphic" creates an odd behaviour

Ion-e, 2025-01-17:
As title, when doing "Alt + click and drag from line graphic" (However, not from one of the extremes of the line graphic) the software behaves oddly.

Couldn't take a screenshot however it's easy to reproduce. 

Actual: It seems a thumbnail of the current data item is created and dragged along, however it disappears when left click is released and instead of dropping the "data item" in a different location (say in  a different display panel) it drags the line graphic to that last position.

Ion-e, 2025-03-21:
Thanks for the explanation!

closed 2025-03-21 by Ion-e

jamesrussell216, 2025-03-26:
I think this (or another issue) needs reopening here; there are weird behaviours that go on with this functionality. For instance if you...
- Press Alt
- Left-click on the middle line
- Drag; this brings up the export interface.
- However if you release the left-click whilst still over the display item, _then move the mouse_, the line profile moves, which I'm fairly sure isn't intended?