nion-software/nionswift #1320: Consistency issue - Selected item in the Data Panel when having a composite line creates is inconsistent to other types of generations

Ion-e, 2025-01-27:
When creating a line composite plot, the created item is not automatically selected, instead it seems like the next item in the data panel list is selected.

This is inconsistent with other types of items generated, where the data item generated is automatically selected.

cmeyer, 2025-03-21:
Can you recheck this after recent data panel changes?

Ion-e, 2025-03-21:
Although not the same behaviour. This seems to still be an issue.

Currently when creating a line composite plot, the selected item would be the one chosen to merge to the second line plot.

It would be arguable to say that the composite line plot should be the item selected just after it's creation.