nion-software/nionswift #1323: Graphic threw error when using it.

Ion-e, 2025-01-29:
Graphic was flipped before this error happened.

[image]

jamesrussell216, 2025-02-20:
@Ion-e can you fill out the repro case for this one please?

Ion-e, 2025-02-21:
Unsure how, but I managed to reproduced it: 

[image]

However, the error seems to be triggered easily when changing the ellipse graphic from 2D to 1D:

As seen in the following image when the ellipse encompases a 2D array of pixels, it is shown like this:

[image]

However, when showing only a 1D array of pixels, it's shown like the following image: 

[image]

This error seems to be thrown when Making the ellipse very very narrow.

jamesrussell216, 2025-03-10:
@Ion-e OK are you basically saying when you define an ellipse that's a 1D array of pixels and do a transpose it throws an error?

cmeyer, 2025-03-20:
I think this is working as expected. The computation is not able to run on data that has a dimension size of 0.

Ion-e, 2025-03-21:
Closed as per comment and resolution. Thanks!

closed 2025-03-21 by Ion-e