nion-software/nionswift #1326: Difficult to select Display item 1D line plot

Ion-e, 2025-01-31:
Observe reproducing this ticket. https://github.com/nion-software/nionswift/issues/1245.

When the line profile of a 1D line plot is wide, it becomes difficult to select the display item (Which may be needed to use the inspector).

jamesrussell216, 2025-03-11:
@Ion-e I'm confused; what's a line profile or a 1D line plot?

Ion-e, 2025-03-12:
Green arrow points towards the line profile graphic.
Red arrow points towards the 1D line plot graph.
Apologies if the terminology is not correct, feel free to correct me (bow).

The issue stated in the bug is that when having a too wide line profile, the selection of the 1D line plot display becomes difficult because the graphic covers most of the item on a higher layer. Hence making it only possible to select the line plot by clicking in the blue square in the image.

[image]

cmeyer, 2025-03-14:
Proposed solution?

cmeyer, 2025-03-17:
See also
- #149 

Overall I think there best solution for this issue (1326) is to click on the lower left corner of the line plot display panel where nothing is drawn. But #149 might work as well.

Ion-e, 2025-03-18:
@cmeyer would that get in conflict with any of the modifier - click functionalities?