nion-software/nionswift #1330: Selected graphics should be unfocused (for inspector) when selecting item in data panel

cmeyer, 2025-02-03:
From PR #1327, @Ion-e:

> 
>     Place a graphic in an item in the workspace.
>     Select the graphic.
>     Change selection to a different data item in the Data panel.
>     Select the previous data item with the graphic in the Data panel.
>     Observe the inspector will only show the graphic's inspector and not the whole data item.
> 
> Please note that this is an inconsistent behaviour from step 5 being performed on a data item in the workspace instead.

Ion-e, 2025-02-25:
when testing this feature. Make sure that the "Go-from-one-display-panel-directly-to-a-graphic-for-one-click-edit-of-graphic" works fine when testing this feature.

Ion-e, 2025-03-28:
Confirmed described issue persists.

[image]