nion-software/nionswift #1336: Add a mechanism for delaying loading a plug-in until a specific resource is available

Brow71189, 2025-02-17:
We already have the option to specify plug-ins that are required before another plug-in can be loaded. This is done in a file called `manifest.json` and is for example used by plug-ins that require `nionswift-instrumentation-kit` such as `nionswift-usim`.
What that system is lacking is a way to specify a specific resource instead of a specific plug-in.

An example for this would be a plug-in that needs to access `stem_controller = Registry.get_component("stem_controller")` during its initialization. Using `manifest.json` doesn't work here, because `stem_controller` can be implemented by `nionswift-usim` or `nion-hardware` or maybe even a third-party plugin. A plug-in using `stem_controller` ideally works with all implementations, so it shouldn't specify a specific plugin in its `manifest.json`, or it will not be able to run with `nionswift-usim` if it requires `nion-hardware` or the other way around.

What would be needed would be a way for a plug-in to require a specific resource to be available before it is initialized. In this example, the plug-in would require `stem_controller` to be available for example from the `Registry`, so its initialization should be delayed until `stem_controller` is available.

cmeyer, 2025-02-17:
You can use code similar to this, although in looking at this code I realize it has a bug: it should not fire "application" events again and should instead call `handle_application_changed` for any already-registered components. Even so, it shows how you can wait for `stem_controller`. Will this work?

https://github.com/nion-software/nionswift-instrumentation-kit/blob/2a8a2396f5b967b3855ca5ad5af504adb025fd99/nionswift_plugin/nion_instrumentation_ui/AcquisitionPanel.py#L1558-L1576

Brow71189, 2025-02-18:
You're right, somehow I forgot about this system. This is about half-way what I'm proposing I would say.

The problem then still is that the plug-in implementor has to delay initialization until that resource is available. That is certainly doable, but it would be more comfortable if Swift would take care of that.