nion-software/nionswift #1340: Having too many daughters/parents for a display item will make them to not show correctly in the display panel

Ion-e, 2025-02-18:
As title and centre display panel (With the RGB).

When an item has too many daughters/parents, the links to that data item (Thumbnail buttons at the bottom corner of that display panel) don't appear in full.

I would expect this tool to have a similar behaviour to that of the Grid/Thumbnail Browser with a "Reel" mode for consistency between similar features.

[image]

jamesrussell216, 2025-02-20:
@Ion-e this is a limitation of the current design however I would imagine it's pretty unlikely that there's a solid use case for that many related items.

Ion-e, 2025-02-21:
Probably!

closed 2025-02-26 by Ion-e